bigjosh/Move38-Arduino-Platform #4: Board Manager URL

jbobrow, 2017-08-30:
`https://bigjosh.github.io//package_index.json` tested with Arduino 1.8.0 and 1.8.4 doesn't add the Move38 Blinks hardware type to the board manager. Looks like the github.io page isn't published

leppert, 2017-10-12:
@bigjosh Any ideas?

bigjosh, 2017-10-12:
Board manager stuff is not rolled out yet, you have to do the local install.

jbobrow, 2017-10-12:
Greg’s up an running with the latest, looks like all of the examples are not working in the Dev branch. Something we’re missing?

––
www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

leppert, 2017-10-12:
```
Arduino: 1.8.5 (Mac OS X), Board: "Blinks Tile"

/var/folders/h5/5rty7brs56l06twmrxkw13s40000gn/T/arduino_modified_sketch_380738/HSBColorWheel.ino:1:19: fatal error: blink.h: No such file or directory
 #include "blink.h"
                   ^
compilation terminated.
exit status 1
Error compiling for board Blinks Tile.
```

bigjosh, 2018-11-16:
Giving up on board manager integration for now. Current system works fine and avoids so many Arduino problems. Maybe someday will will make a portable library stub that comes over the Arduino board manager.

closed 2018-11-16 by bigjosh