bigjosh/Move38-Arduino-Platform #26: Bright Red LEDs after unknown period of time/sleep

jbobrow, 2018-01-16:
Tends to light up not all pixels, but 4 or 5 of them very brightly. Button presses have no effect, and reset is necessary. Similarly, sometimes Blinks will remain in a state with no display at all, i.e. dark. This seems to happen with the same frequency as waking into a state of bright red, possibly a little more often.

bigjosh, 2018-01-16:
Wow, I've never seen anything like this. Does this happen spontaneously while they are asleep, or after they wake from sleep by button press? Which game does it happen on? Several?- All?

jbobrow, 2018-01-16:
I believe I see it on Fracture after wake

All my best,
Jonathan

––
www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

bigjosh, 2018-01-17:
This could be a manifestation of the challenges of writing correct code when loop() can be interrupted by sleep at any time. Unless you feel strongly, rather than spend time hunting this down I would rather get the new non-preemptive sleep working and then see if that resolves the issue. Good?

jbobrow, 2018-01-17:
Feels appropriate, I think getting the model right makes sense. Proceed :)

All my best,
Jonathan

––
www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

bigjosh, 2018-01-25:
Kick? Still an issue?

jbobrow, 2018-01-25:
Not certain, code base has been changing so much and the bug wasn't clear if it was something that showed up after 15 min, an hour, or a couple of days.

I’ll set aside some blinks with new sketches later today and see how they perform over the course of the next couple of days.

––
Jonathan Bobrow
CEO & Co-Founder
Move38

sign up here to learn about our upcoming kickstarter campaign

www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

jbobrow, 2018-10-25:
Seems like this is an artifact of the outdated version that used interrupts

closed 2018-10-25 by jbobrow