bigjosh/Move38-Arduino-Platform #32: DEV branch - makeColorRGB() should be 0-255

jbobrow, 2018-01-22:
Looks like makeColorRGB() is currently using values between 0-31… expected behavior is 0-255 (even if the values are being scaled/mapped).

bigjosh, 2018-01-22:
0-31 is the documented behavior...

```
// We inline this so we can get compile time simplification for static colors

// Make a new color from RGB values. Each value can be 0-31. 

inline Color makeColorRGB( byte red, byte green, byte blue ) {
    return MAKECOLOR_RGB( red , green , blue );
}
``

https://github.com/bigjosh/Move38-Arduino-Platform/blob/944ac260d6c0650e72039bfef79f1f536153dddf/libraries/blinklib/src/blinklib.h#L136

... and I think the correct one since it does not imply that you have more precision than actually exists.

It should actually be documented as "Each value can be 0-`MAX_BRIGHTNESS`". 

But mostly a style decision (although current model is slightly more performant for non-constant values), so whatever you think fits programmers' brains better.

jbobrow, 2018-01-23:
Please keep it as previously discussed 0-255 facing the developers. I can note in the documentation that the hardware does not differentiate between values such as 253 and 254 and such subtlety should not be relied upon for meaningful game state.

I prefer to keep users familiar with the 8-bit color system that is easy to reference in any design program or convert from a hex value that someone likes on the web.

The desired behavior is that no matter what we scale the brightness values to on the backend, the front end always treats 0-255 as darkest/off to brightest/on.

––
www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

bigjosh, 2018-01-23:
So you want the color returned by `makeColorRGB( 7 , 7 , 7 )`  to be `OFF`?

-josh

bigjosh, 2018-01-23:
Should the `brightness parameter in...

```
// Dim the specified color. Brightness is 0-31 (0=off, 31=don't dim at all-keep original color)
// Inlined to allow static simplification at compile time

Color dim( Color color, byte brightness);
```
... now also be in the range 0-255 to match?

closed 2018-01-23 by bigjosh

jbobrow, 2018-01-23:
Btw, values should round up… i.e. makeColorRGB( 1 , 1 , 1 ) should be the dimmest of dim. And makeColorRGB( 248 , 248 , 248 ) should be the brightest of bright.

0 is the only off value.

Best,
jb

––
www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

bigjosh, 2018-01-24:
> Btw, values should round up… i.e. makeColorRGB( 1 , 1 , 1 ) should be the dimmest of 
>dim. And makeColorRGB( 248 , 248 , 248 ) should be the brightest of bright.

Implemented in 5ee4b6d36feecb803cc1715be4582a0ce7d9e543 .

Slightly non-linear since 31 does not go into 254 evenly, so for example RGB values of 1 and 9 both map to brightness 1.

commit c73b79b8 (2018-01-23): MakeColorRGB() now takes values 0-255 rather then 0-31

Still only 31 actual brightness levels, so just an abstraction for the programmer view. Closes #32