bigjosh/Move38-Arduino-Platform #46: Animation Library Problems

JustinHaGames, 2018-07-27:
I was creating a Blink that can demo all of the animations from the animation library and came across 2 problems. 

Flash(RED,500);

Flash makes a color show up but it lasts longer than the duration. This is probably due to the fact that Flash() doesn't know what the original color was so it does not switch back. 

fadeTo() doesn't seem to be implemented.

bigjosh, 2018-07-27:
Sorry, the the animation stuff is not ready for primetime. Feel free to look at it and play with it if you want, but these functions are not part of the API yet, and the final animation library will likely be different. Thanks!

bigjosh, 2018-11-16:
Animation is not going to be part of the core platform. Instead it will be offered in optional libraries.

closed 2018-11-16 by bigjosh