bigjosh/Move38-Arduino-Platform #55: Not able to transfer games

jbobrow, 2018-11-11:
The current Dev branch compiles and runs, but with some oddities.
1. button presses don't seem to work.
2. after holding for 3s, the Blink flashes white and then returns to its display state
3. no transfer begins

bigjosh, 2018-11-11:
Can you send me the HEX file that Arduino leaves in the build directory and a copy/paste of the text output from the build?

bigjosh, 2018-11-11:
When you push "upload" in the IDE, do you notice that it sends two HEX files - one for the sketch and one called `bootloader.hex`?

jbobrow, 2018-11-11:
Ah, it looks like it can’t find the boot loader binary.

Here is the output from my upload:
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/jonathanbobrow/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/jonathanbobrow/Documents/Arduino/libraries -fqbn=Move38-manual-Josh:avr:blink -ide-version=10807 -build-path /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220 -warnings=all -build-cache /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_cache_913024 -prefs=build.warn_data_percentage=75 -verbose /Users/jonathanbobrow/Documents/Arduino/Blinks Games/Berry/berry.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/jonathanbobrow/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/jonathanbobrow/Documents/Arduino/libraries -fqbn=Move38-manual-Josh:avr:blink -ide-version=10807 -build-path /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220 -warnings=all -build-cache /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_cache_913024 -prefs=build.warn_data_percentage=75 -verbose /Users/jonathanbobrow/Documents/Arduino/Blinks Games/Berry/berry.ino
Using board 'blink' from platform in folder: /Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr
Using core 'blinkcore' from platform in folder: /Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega168pb -DF_CPU=8000000L -DARDUINO=10807 -DARDUINO_AVR_BLINK -DARDUINO_ARCH_AVR -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/cores/blinkcore -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/variants/standard /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220/sketch/berry.ino.cpp -o /dev/null
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega168pb -DF_CPU=8000000L -DARDUINO=10807 -DARDUINO_AVR_BLINK -DARDUINO_ARCH_AVR -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/cores/blinkcore -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/variants/standard -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinklib/src /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220/sketch/berry.ino.cpp -o /dev/null
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega168pb -DF_CPU=8000000L -DARDUINO=10807 -DARDUINO_AVR_BLINK -DARDUINO_ARCH_AVR -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/cores/blinkcore -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/variants/standard -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinklib/src -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinkos/src /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220/sketch/berry.ino.cpp -o /dev/null
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega168pb -DF_CPU=8000000L -DARDUINO=10807 -DARDUINO_AVR_BLINK -DARDUINO_ARCH_AVR -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/cores/blinkcore -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/variants/standard -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinklib/src -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinkos/src /Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinklib/src/Print.cpp -o /dev/null
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega168pb -DF_CPU=8000000L -DARDUINO=10807 -DARDUINO_AVR_BLINK -DARDUINO_ARCH_AVR -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/cores/blinkcore -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/variants/standard -I/Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/libraries/blinklib/src -I/Users/j [... cut ...]
avrdude: load data flash data from input file /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220/berry.ino.hex:
avrdude: input file /var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220/berry.ino.hex contains 3760 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 3.75s

avrdude: verifying ...
avrdude: 3760 bytes of flash verified

avrdude done. Thank you.

––
Jonathan Bobrow
CEO & Co-Founder
Move38

www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

jbobrow, 2018-11-11:
**Bootloader file specified but missing: /Users/jonathanbobrow/Documents/Arduino/hardware/Move38-manual-Josh/avr/bootloaders/bin/blinkboot.hex

This feels like an easy fix. When should that be generated?

––
Jonathan Bobrow
CEO & Co-Founder
Move38

www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

jbobrow, 2018-11-11:
Do I need to Burn Bootloader first?

Still seeing the same behavior even with the bootloader in the correct path

––
Jonathan Bobrow
CEO & Co-Founder
Move38

www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

bigjosh, 2018-11-11:
This is the same issue with the IDE not following the "upload" recipe from the `platform.txt` for the board.  There must be some hard coded rules deep in the uploader code that decide when to follow and when to make it up, because the steps above do not match the ones in the `platform.txt` (you can [check for yourself](https://github.com/bigjosh/Move38-Arduino-Platform/blob/dev/platform.txt#L109)), and those rules seem to end up being OS dependant? I can also tell it is not running the pattern becuase the avrdude does not have the `-B 4` in the middle , so the IDE is getting the pattern it is using from somewheres else or making it up. 

I have noticed, for example, that if you do not specify an upload baud rate in the `boards.txt`, then the IDE does not run the `upload` pattern in the `platform.txt` even if that pattern does not use the baud rate. So there is someplace where it has these rules, and the alternate patterns that it throws in. 

Do you want to try digging in the source to find these rules? 

Can you try the "upload using programmer" menu choice instead?

If not, you can open a command window in the sketch working directory (`"/var/folders/82/pl2rlhl125zdyh4dbwz52tkc0000gn/T/arduino_build_399220/` in your case) and run the `avrdude` command by hand., which should be...

```
avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf --B 4 v -patmega168pb -cusbtiny -Uflash:w:berry.ino.hex:i "-Uflash:w:{runtime.platform.path}/bootloader/bin/blinkboot.hex:i"
```

... where `runtime.platform.path` is the path where the `platform.txt` lives.

bigjosh, 2018-11-11:
"*Bootloader file specified but missing:" this is part of the Arduino "burn bootloader" system, which we are not using and so silly it should print this. I deleted the reference in the `boards.txt` just to get rid of the message, but this is not an issue.

bigjosh, 2018-11-11:
Ok, just pushed commit fc909e1 that adds a duplicate `programmers.txt`. Pull this and restart the IDE and then pick the &copy* of your current programmer like so...

[image]

...and then try uploading again. You can know if it worked if you see the `-B 4` on the `avrdude` command line.

jbobrow, 2018-11-11:
Still not functioning as expected from in the IDE. The workflow that has worked for me, sort of, is exporting the binary, which goes to the sketch’s folder and then using AVRDude from the command line to install the file that is file-name.withbootloader.hex.

I did that with a couple of examples and installed them. They seem to have transmitted once successfully, but haven’t had a second successful transmission :/

––
Jonathan Bobrow
CEO & Co-Founder
Move38

www.troxes.com | www.move38.com | www.jonathanbobrow.com | www.bitwise-design.com

jbobrow, 2018-11-15:
This is functioning as expected, when using the correct programmer

closed 2018-11-15 by jbobrow