bigjosh/Move38-Arduino-Platform #59: Individual Gamma table for RGB

jbobrow, 2018-11-28:
White at the low end is heavily skewed in the Blue direction, having individual gamma tables seems appropriate for our hardware but takes up another 64 bytes, if I understand this correctly.

bigjosh, 2018-11-28:
Correct - right now we have a single table of 32 bytes that lives in the bootloader area. All three R,G,B currently use the same table. We could break that out into individual tables for each color, or one table for R & G and another for just B. Or if there is just one or two RGB values of white that look off, we could special case those out.

bigjosh, 2019-01-25:
Not gonna happen - no room. We can possibly slightly tweak individual values if anything is really ugly, but looks pretty smooth to my eye now!

closed 2019-01-25 by bigjosh