bigjosh/Move38-Arduino-Platform #65: throw out the buttonPressed() buttonDown() buttonReleased() flags on the wake

bigjosh, 2019-01-22:
can we throw out the buttonPressed() buttonDown() buttonReleased() flags on the wake? i.e. when I wake up a game of Berry, it currently advances to the next color, this means that we essentially wait for button released to happen once before it is accessible... I see how this can be a little murky, but think it is weird to have wake actions drive game actions.

bigjosh, 2019-01-25:
@jbobrow I think this is more complicated. 
So if blink naturally falls alseep with the button up, then wakes because of a button press. The game starts and the button is still currently down, but you do not want `buttonDown()` to return true in this case? If the button continues to be held down long enough for a long press, do you then want to also suppress that event (which is itself a little complicated)? If not, then I also think it is incoherent to get a long press event while the button seems to be up. 

I imagined that games like Berry could use the isWoke() test to see if they just woke, and if so then not interpret the current down like a down during normal running. 

Alternately we could make a rule that when you press to wake the game does not actually start until you *release* the button, so the button is always guaranteed to be up when a game starts. In this case we maybe want to show the new wake animation for as long as the button is held down (or at least until it is held down long enough to trigger a seed or sleep)?

jbobrow, 2019-04-01:
I think the isWoke() test is the right way to handle this. Let's test it with Berry. It is the perfect example.

closed 2019-04-25 by bigjosh