bigjosh/Move38-Arduino-Platform #68: Two blink red on failed program download.

bigjosh, 2019-01-25:


jbobrow, 2019-04-06:
Blink does a two red pulse flash until one of two things happens. 1) A piece tells it to go into listen/program mode, in which case it listens to the programming. 2) User touches the top and it quits out into its own game.

bigjosh, 2019-04-06:
This is much more complicated (that is, flash heavy) to do correctly that it might seem. To be able to do above behavior, the red blinks would actually need to be a little game, and that game would need to be kept in flash and downloaded to active are when a download fails. Alternately we could also include a hacked "failed download" mode in every real game. 

Does not seem worth it for this? Could, say, 2 sets of of two 2 blinks followed by a jump into the built-in game be good enough?

jbobrow, 2019-04-07:
I know that people will be failing to upload games consistently, the first thing I see people do is grab them to hold them and feel them during game transmission. This happens even after explaining to wait for the game to finish loading. This makes me feel strongly that this interface needs to feel as good as possible, otherwise game players will start the game very confused. Let's talk this one through some more, I am not opposed to the failed download mode being in every game, it does feel hacky, but sounds like it might be the best way to deliver the experience I am looking to deliver.

bigjosh, 2019-04-25:
Currently 3 red blinks. After failed download, blink can immediately accept a new download seed, or a button press restarts the blink with the built-in game.

commit 2eabbcff (2019-04-09): Pul in new failed download mode in BIOS

on failed download will blink twice red until either button press (which starts builtin game) or new seed packet received (which restarts normal download proceedure).

Closes #68
closed 2019-04-25 by bigjosh