bigjosh/Move38-Arduino-Platform #72: Animation on Wake

jbobrow, 2019-01-25:
Allow for a half second white spiral w/ trail on wake

jbobrow, 2019-01-25:
30ms on each LED with white increasing from 0 - 31

bigjosh, 2019-01-25:
So with [~16 steps](https://www.google.com/search?q=0.5s+%2F+30ms&oq=0.5s+%2F+30ms&aqs=chrome..69i57j6.1998j0j9&sourceid=chrome&ie=UTF-8)?

jbobrow, 2019-01-25:
I should just try it out, and send the exact description

jbobrow, 2019-04-07:
Wake and sleep animations are detailed in issue #73, that is the one to follow :)

jbobrow, 2019-04-22:
Much better now, however, the animation should be white for the wake. The travel from Tile to tile is much quicker than the sleep propagation, matching would feel the best, i.e. slowing the tile to tile on wake, however, it is not worth losing memory for such an update.

jbobrow, 2019-05-08:
@bigjosh just another ping on the animation for wake to be white instead the current blue

commit 906a3b34 (2019-04-09): Sleep and wake animations

Closes #73 and closes #72
commit 241225ce (2019-05-10): Change warm wake animation color from blue to white

CLoses #72
closed 2019-10-30 by jbobrow