bigjosh/Move38-Arduino-Platform #73: Animation on Sleep

jbobrow, 2019-01-25:
Blue spiral in the reverse direction of the wake spiral for half a second (fade down into darkness

bigjosh, 2019-04-06:
Can you be more specific? Code example?

jbobrow, 2019-04-06:
I sketched out something that has a rotational [wake and sleep](https://github.com/Move38/SketchADay/blob/master/004_WakeSleep/004_WakeSleep.ino), but think that it will look better if all 6 LEDs perform the same since it will be across a cluster of Blinks and different orientations will make the rotation look less elegant.

To be simpler, the wake can simply be a 255 -> 0 quadratic decay (WHITE). Sleep can be a 255 -> 0 (BLUE) decay of the same.  200ms in duration.

bigjosh, 2019-04-06:
Example sketch for how you'd like both effects to look?

jbobrow, 2019-04-07:
I feel good about this. Copy these animations :)

0->200 brightness WHITE for wakeup
200->0 brightness BLUE for sleep

300ms duration for both.

```
/*
 * Wake/Sleep Animation
 */
#define DURATION 300
#define MAX_BRI  200

bool isAwake = false;

Timer wake;
Timer sleep;
 
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  if(buttonPressed()) {
    isAwake = !isAwake;
    if(isAwake) {
      wake.set(DURATION);
    }
    else {
      sleep.set(DURATION);
    }
  }

  if(isAwake) {
    setColor(GREEN);
  }
  else {
    setColor(OFF);
  }
  // sleep
  // quickly fade up to blue
  if(sleep.getRemaining() > 0) {
    setColor(dim(BLUE, map(sleep.getRemaining(),0,DURATION,0,MAX_BRI)));
  }

  // wake
  if(wake.getRemaining() > 0) {
    setColor(dim(WHITE, MAX_BRI-map(wake.getRemaining(),0,DURATION,0,MAX_BRI)));
  } 
}
```

bigjosh, 2019-04-09:
So here the `setColor(GREEN);` would be replaced with just whatever the game does when woken, right?

jbobrow, 2019-04-09:
Yeah, exactly

commit 906a3b34 (2019-04-09): Sleep and wake animations

Closes #73 and closes #72
closed 2019-10-30 by jbobrow