bigjosh/Move38-Arduino-Platform #74: Timing for Blinks Button Actions

jbobrow, 2019-04-03:
1200ms - for buttonLongPressed()
3000ms (alone) for program mode (blue eratic)
6000ms - sleeper hold (spread sleep)

bigjosh, 2019-04-04:
Length of long button press decreased from 2000ms to 1200ms in the blinkBIOS repo commit 72aea1f703938a4fc97c585dac4fa628170175d8.

bigjosh, 2019-04-04:
Updated names and timings for 6 second and 7 seconds button presses to 3 seconds and 6 seconds respectively n the blinkBIOS project commit 609dbb3198814368b6d3dcfba3ff8cb90d0a100c.

bigjosh, 2019-04-04:
@jbobrow Needs to be tested and then closed!

bigjosh, 2019-04-06:
This is now implemented in the `/dev` branch with 4s for seed mode and 6s for sleep. 

LMK if you want to switch back/back to 3s for seed mode!

commit b96abb85 (2019-04-06): Oops, that was supposed to be 4s rather than 3s

3s did seem too short. Closes #74
closed 2019-04-06 by bigjosh