bigjosh/Move38-Arduino-Platform #78: sendDatagramOnFace command needs to reverse order

ReluctantPirate, 2019-04-18:
Currently, the command sendDatagramOnFace has the internal structure of (face, data, size). In other commands with the word "face" in them, the internal structure matches the name.

To make this command conform, the internal structure should be:
sendDatagramOnFace(data, size, face)

bigjosh, 2019-04-18:
I agree! This confuses me all the time!

I'd argue that `face` should always be the first param in all signatures like `setFaceColor()`  to match the "destination first" convention that appears everywhere in C-like languages from `fopen()` to `memcpy()`.

But no matter what, @ReluctantPirate you are right it should be consistent throughout the API. 

@jbobrow Thoughts on which way to go? I'm guessing that you'd rather make the datagram class of functions _wrong_ than make all the others right at this point since almost no code dependant on the datagram stuff yet?

jbobrow, 2019-04-18:
@bigjosh, I believe it was the _correct_ naming of `setValueSentOnFace` that started the pattern. Since we didn't have a nice sounding name to bring face to be the first parameter in that case, we made the move to `setColorOnFace`. In theory, that is also nicer for autocomplete since `setColor` and `setColorOnFace` show up together. I think that `sendDatagramOnFace()` is the way to go, and I paused to think of `sendDatagramOfLengthOnFace()` but think it is unnecessarily long. Lets go with `sendDatagramOnFace()` and stay consistent.

commit eca6dfbf (2019-04-20): Make `SendDatagramOnFace()` arg order consistant with `SetColorOnFace()`

Closes #78
closed 2019-05-10 by bigjosh