bigjosh/Move38-Arduino-Platform #80: IR warm sleep wake neighbor time to tile timing

jbobrow, 2019-04-20:
Previous versions seemed to have a 50ms or shorter wake time from a warm sleep. It appeared as though the waking of neighbors happened as quickly as possible. The latest dev has a delay that looks like the Blink animates and then wakes it’s neighbor. The wake should travel quickly so that the tile to tile animation looks closer to the pace of tile to tile in the force sleep.

bigjosh, 2019-04-20:
You have two choices: Show the animation then send the wake packets, or send wake packets and then show the animation. 

I picked animate then send because (1) it feels more responsive from UI point of view, (2) it adds a nice little visual delay when the wake wave propagates and I even thought I remembered to saying at some point that you wanted to add a delay here anyway.

Alternately you could still do the animation first but just make the animation faster if it feels like it all takes too long now.  

So LMK what you want!

bigjosh, 2019-04-20:
...or potentially we could try to multitask the sending and the animating, but seems like too much complexity and space for this?

jbobrow, 2019-04-20:
I agree with the motivation, it would be great to have the passing of the wake happen 1/4 or 1/2 of the way into the animation, this way the white wake animation passes from blink to blink rather than a gameplay display then causing the wake animation.

bigjosh, 2019-04-20:
Sending IR packets takes time and the wake sequence sends lots of them. If we just stopped in the middle of the wake to send them ,then it would not look good because of the big pause in the middle.

I can play around with staggering the packets throughout the wake sequence if it is important.

bigjosh, 2019-04-21:
Only because I love you, I just committed 8946069842998622aee2c38451a2b3eebde70eda which multitasks the animations and the packet sending. Because everything is hard coded, it is actually 40 bytes smaller. 

I think it looks pretty sweet. LMK what you think. The timing is just how things happened to end up with the packets we were all ready sending.  If you want the animations to take longer then we can potentially send extra superfluous packets to stretch things out.

bigjosh, 2019-04-25:
@jbobrow Good now?

jbobrow, 2019-10-30:
Whoops, on the lack of response to these comments. I am sure I already said so, this stuff is wonderful! Closing issue :)

closed 2019-10-30 by jbobrow