bigjosh/Move38-Arduino-Platform #85: Fix mortals example

bigjosh, 2019-04-25:
Needs to be updated to latest API and checked?

bigjosh, 2019-04-25:
@jbobrow Do you have a working version of Mortals that we can put here? Is this the right game to ship with the API as an example? I think we should give at least one full blown game for people to look at.  Should we ship all the games as examples?

jbobrow, 2019-04-25:
I think fracture and mortals belong at very least

––
Jonathan Bobrow
CEO & Founder
Move38
Pronouns: he/him

www.move38.com | www.jonathanbobrow.com

bigjosh, 2019-04-25:
Do you have working versions of these that you can PR into this repo?

jbobrow, 2019-04-25:
I’ll try and get to this tonight tonight, I know I branched for the compression recently.

––
Jonathan Bobrow
CEO & Founder
Move38
Pronouns: he/him

www.move38.com | www.jonathanbobrow.com

jbobrow, 2019-05-08:
After the RC is pulled into the Move38 Repo, I'll add the completed games to the examples folder and then you can pull them back into your repo :)

bigjosh, 2019-05-10:
Removed,

closed 2019-05-10 by bigjosh