bigjosh/Move38-Arduino-Platform #88: Message received in game on game transmission completed

jbobrow, 2019-05-13:
This has been a tough one to understand even though we can recreate it every time. The best way to see this is here in [Speed Racer](https://github.com/Move38/SpeedRacer/tree/newArchitectureDK). If you put the game on ~3 individual Blinks, they will be in a ready cyan state, then they recognize neighbors with yellow and then a double click of a single one will pathfind the road and start the game. If a game is started and a tile is alone, it will be magenta. If a tile is next to an in game tile, it will also start the game, so if a magenta tile is next to a cyan tile, it will turn it magenta.

The problem is that at the end of game transmission, we expect all of the Blinks to be in cyan (with yellow showing on neighbors). The current result is that almost immediately out of the game transmission, a message is received that the game is in play. Could this be received from the neighbor that hasn't yet popped out of program mode? That seems like the culprit.

bigjosh, 2019-05-13:
Sorry, I do not understand. Can you make napkin state diagrams for both intended and observed operation?

jbobrow, 2019-05-13:
Will do

––
Jonathan Bobrow
CEO & Founder
Move38
Pronouns: he/him

www.move38.com | www.jonathanbobrow.com

bigjosh, 2019-05-13:
I added code to print all state changes to serial port (see PR), and the game seems to always start in `SETUP` mode after a download, and then I can change to `PATH` with a double click.  What is supposed to happen?

bigjosh, 2019-05-13:
OK, I think I see what you mean. This line is getting executed right after the game starts after a download...

`   } else if (getGameState(getLastValueReceivedOnFace(f)) == PLAY) {//transition to PLAY`

I will track down what is going on later today.

jbobrow, 2019-05-13:
That's correct, however, when it completes the game transmission, it goes into `SETUP` and then directly into `FREEAGENT`

jbobrow, 2019-05-13:
Here is a mockery of a state diagram, but should show the issue [image]

bigjosh, 2019-05-13:
OK, I will take a look this afternoon!

jbobrow, 2019-05-13:
My guess is that there is a message/value lingering on a face when they enter the game from transmit.

jbobrow, 2019-05-13:
Thanks Josh!

This is top priority for me to get the FW in so Jetta can start the
samples. Aiming to have the large mass of Blinks available for E3, which is
only 4 weeks away (yikes).

As always, feel free to ping.

best,
jb
ᐧ

bigjosh, 2019-05-14:
OK, this was a hard one to figure out!

It turns out that the timing in the case is *exactly* right to have the `LETS GO` packet that is being transmitted by one blink after the download is complete collide with the `USER DATA` packet being transmitted on the other blink. The way they collide causes a couple bits at the end of the user packet to get flipped so now rather than receiving a value of 0x00 as transmitted, it gets the mangled value 0xea.  If you look at the bits of 0xea, you'll see it has a `2` in the `state` field of the SpeedRacer packet, so upon getting this the SpeedRacer program goes into play mode, so it starts sending play mode, which put the other tile into play mode. 

Here is the trace of the exact moment when it happens...

[image]

The top trace is the first blink sending a `USER_DATA` packet with the value 0x00, the bottom trace is the second blink sending a `LETS_GO` packet just a smidgen after the first packet had started (a little later and it would have seen the incoming packet start and aborted the send).

Normally this would be no problem because the clobbered received packet would have failed our hamming test, but to make room for viral stay awake, we had to [drop the hamming test and go for a simple parity check](https://github.com/bigjosh/Move38-Arduino-Platform/issues/66#issuecomment-482311743) - which only catches an odd number of bit flips (in this case it was an even number). 

So this is the first case where dropping error detection meant that we got error data- and we knew it was coming.  We still have the same trade off. We can take more bits for error detection, leaving fewer bits for games to send or we can increase the number of bytes sent, which decreases communication throughput. 

I do think I can prevent this very specific case by putting in code that basically aborts any receive in progress every time we transmit a bit. This would mean that in cases like this, nothing would have been received at all.  I don't think this will cause any problems, but we will need to test a lot to be sure.  I'll try to get this done today.

jbobrow, 2019-05-14:
This makes perfect sense, glad it showed up here. To me it seems like a little bit of latency (are we talking tens of ms? or tenths of a second?) every time the button is pressed is a fair trade off and then keeping the 6 bits of communication during gameplay and having the 2 bits for hamming is a good way to go. What is the work involved in making this kind of change, feels like communication changes like this are not simple changes.

bigjosh, 2019-05-14:
...so I just want us all to take a moment to appreciate the requisite timing that it took to produce this collision.   There is [code in place](https://github.com/bigjosh/Move38-BlinkBIOS/blob/master/src/blinkbios_irdata.cpp#L432) that prevents a TX from starting if there is already an RX in progress, but in this case the 2nd packet began *inside* the sync pulse of the 1st packet. If it had happened a few usecs earlier, then the 2nd packet would have been suppressed and it had started a few usecs later then it would have been aborted.  Just to put into perspective how sparse these channels are...
[image]

bigjosh, 2019-05-14:
/dev now includes an updated BIOS that adds an atomic block prevents an RX sync from being received during the window that starts when the decision is made that it is clear to send until after the send is completed.

This cures the collision in this specific case and ones like it. I can't think of a case when this would have negative side effects, but it is a significant change to IR behavior so be on the lookout for anything new and unexpected. 

This costs 6 bytes of RAM and a few dozen bytes of flash. It should not cost any IR bandwidth since any RX that is blocked by this new semaphore would likely have gotten garbled anyway.

jbobrow, 2019-05-15:
Hmmm. I'm still seeing the same behavior at the start of the game. Restarted Arduino and double checked the repo just to make sure I wasn't uploading from the wrong one.

bigjosh, 2019-05-15:
Seemed to work for me, so I must have missed something. Will take a look tomorrow!

bigjosh, 2019-05-16:
Please try again!

I just fixed two bugs on /dev, one of which was responsible for this still happening.

Hopefully all fixed now, but please test and LMK!

(Other was obscure race condition that was introduced moving things around to get flash to fit, but now we have room so moved it back.)

jbobrow, 2019-05-16:
This is working and looks good to me. I'll let you know if I see anything funny arise, but so far so good.

commit 0252e251 (2019-05-14): Add semaphore to block RX durring TX

Update BIOS to version that includes atomic semaphore.
https://github.com/bigjosh/Move38-BlinkBIOS/commit/68c977ab27f1ab25831a1b248018c7070e55bdf1
Closes #88
closed 2019-05-16 by jbobrow