bigjosh/Move38-Arduino-Platform #89: Can

jbobrow, 2019-05-17:


jbobrow, 2019-05-17:
Would like to handle some situations onWake of a blink and couldn't find a flag in the API that allowed for this (most likely just missing it)

bigjosh, 2019-05-17:
`[hasWoken()](https://github.com/bigjosh/Move38-Arduino-Platform/blob/dev/cores/blinklib/blinklib.h#L346)`

...but I am warning you that the semantics are mushy. [We never pinned down](https://github.com/bigjosh/Move38-Arduino-Platform/issues/58) things like is fist power up after battery insert a wake? Is starting up after a successful download a wake? Several others, I think no right answers but should at least be thoughtful and coherent - which it is not right now.

closed 2019-05-17 by bigjosh

bigjosh, 2019-05-17:
Keep in mind that this function is part of blinklib so each game will always see the same semantics it was compiled against, so we can refine semantics any time and new games get the benefit.