bigjosh/Move38-Arduino-Platform #90: map function high to low

jbobrow, 2019-05-30:
Currently the `map()` function works with the following:
`map(var, low_input, high_input, low_output, high_output)`

i.e. lo,hi,lo,hi

I would like to have the function also work for the following:
lo,hi,hi,lo
hi,lo,hi,lo
hi,lo,lo,hi

Some common use case examples are mapping the remaining time from a timer which is going from high to low or wanting to dim brightness with the increase of a variable... All in all these are conveniences, but I have found that only working properly for the lo,hi,lo,hi makes it easy to make mistakes the the 3 other possible configurations. Working with `words` still makes sense, no need to account for negative numbers, I think that is a fine place to draw the line.

bigjosh, 2019-06-01:
Of course it is easy to make a `map()` function that covers these additional cases, but these will expand the flash used by this already-too-big function by several times. 

Maybe make a new `expensive_map()`  function that does not require the arguments to be in min < max order and instead covers all the combinations? 

Or maybe we could change the type of the inputs to be `int`s so that the function could use signed math rather than special casing out the combinations?

jbobrow, 2019-06-02:
Using int makes a lot of sense, I’m trying to think of the main consequences of that change. Ints are twice the size, but I imagine the math is the expensive portion. This isn’t prohibitive for our launch games, but trying to make the platform friendlier for beginners.

bigjosh, 2019-06-02:
Both `int` and `word` are 16 bits on this platform, only difference is that `int` uses one of those bits for sign so goes from -32768 to 32767 whereas `word` goes from 0-65535.