bigjosh/Move38-Arduino-Platform #98: Waking button press incorrectly discarded on blink that was manually put to sleep

bigjosh, 2020-08-25:
From [user BGA on the forums](https://forum.move38.com/t/correct-way-to-deal-with-button-click-to-wakeup/579/7):

I think I found the problem and it looks like a blinklib bug.

The code paths for sleeping when you are the Blink being long-pressed does this:

```
if (blinkbios_button_block.bitflags & BUTTON_BITFLAG_6SECPRESSED) {
        // Held down past the 7 second mark, so this is a force sleep request

        warm_sleep_cycle();

        // Clear out the press that put us to sleep so we do not see it again
        // Also clear out everything else so we start with a clean slate on
        // waking

        blinkbios_button_block.bitflags = 0;

}

```

While the code path for when it goes to sleep because it was told to do so by another Blink does this:

```
if (packetDataLen == 2 && decodedByte == TRIGGER_WARM_SLEEP_SPECIAL_VALUE && packetData[1] == TRIGGER_WARM_SLEEP_SPECIAL_VALUE) {
                warm_sleep_cycle();
}

```

Note the missing blinkbios_button_block.bitflags = 0 in the second case.

I added it to my local blinklib and now it works as expected.

brunoga, 2020-08-25:
Juts to clarify, there is something wrong but I suspect my workaround only worked by chance and in my specific case but the actual issue is a different one. If you try this simple program:

```
#include <blinklib.h>

void setup() {}

void loop() {
  bool click = buttonSingleClicked();
  bool woke = hasWoken();
  if (click) {
    if (woke) {
      setColor(RED);
    } else {
      setColor(GREEN);
    }
  }
}
```
It currently fails if you click the same Blink that put others to sleep (the actual click is discarded so assuming the Blink was with color OFF, it will not change to either RED or GREEN).

Now if you click a Blink that was not the one that initiated the wakeup, It turns GREEN on wakeup (I was expecting it to turn RED).

If I do the opposite of what I did and *REMOVE* blinkbios_button_block.bitflags = 0 when it is called after warm_sleep_cycle() returns, then clicking on the Blinbk that started the sleep works as expected (it will turn RED), but clicking on other blinks will not work (they will turn GREEN while I woudl expect RED).

After investigating more, I think I found the actual fix There are 2 issues:

1 - Button clicks are consumed when they should not.
2 - Button click status is updates *BEFORE* RX_IRFaces run. This is where the Blink stops when asked to sleep by another Blink so when it returns the button clicks are being reset and are lot.

The fix is remove all  blinkbios_button_block.bitflags = 0 when it is called after warm_sleep_cycle() and moving RX_IRFaces() to run before the button clicks are updated.

brunoga, 2020-08-25:
See https://github.com/brunoga/blinklib/commit/e311203084db197f96b79826b52be6cbcbaa8127

PR #99 (2020-09-13): Fix button click detection after sleep.
- Do not clear button bit flags after warm_sleep_cycle().
- Move RX_IRFaces() to run before button state is reset.

Fixes #98.
closed 2020-09-13 by bigjosh