bigjosh/Move38-Arduino-Platform #104: Datagram example magic value doesn't appear to be used

jbobrow, 2021-03-26:
I think this code intends to set the magic value when a datagram is confirmed to be received completely, but right now, it looks like it is always sharing that magic value. This could be a misunderstanding on my part as well.

jbobrow, 2021-03-26:
'F-SimpleDatagramDemo' for reference

bigjosh, 2021-03-27:
I believe the code is as intended, which might be overly complicated for this example. 

The idea is to show the difference between how face values and datagram messages work (face values are continuously sent while datagrams are sent once). The `42` is set as the face value on all faces so it is sent continuously whenever a datagram is not being transmitted. On the receiving side, a `42` is shown as green to indicate that we successfully got the intended face value value from the other side, just as we show blue if we successfully got the intended datagram value from the other side ("CAT"). 

Make sense? 

Now all those misspellings are definately inexcusable! :)