bigjosh/Move38-Arduino-Platform #105: Dev Branch 328 Communication bug (ignored faces)

ReluctantPirate, 2021-07-14:
We have experienced a confusing set of bugs and behaviors with the new dev branch code. None could be reliably reproduced, but I'll describe them below:

1. Playing with a set of Blinks with mixed codebases (many using production code, some using the 168 code within the dev branch), there were faces that simply ignored each other. While playing this game - https://github.com/ReluctantPirate/OnOff/tree/multiColor - certain connection points between 2 Blinks would be ignored - they simply do not see or acknowledge each other. Rotating and rearranging the Blinks made the problem go away, and it didn't seem to be the same Blink or Blinks over and over - it just happened randomly.

2. While testing Hexenwood_328, we created a setup that had a single connection point that suffered from this - the Blink was not part of the information wave used to start the game (it only had a single face connected to the cluster, which was ignored by one or both sides). Both of the Blinks involved in the connection were production Blinks taken straight from a sealed game system, but running the Hexenwood_328 code.

As we continue this testing, I will try to capture examples of this.

bigjosh, 2021-07-15:
Can you try testing the various combinations using the IR tester program rather than games? 

It would be helpful to know which combinations the failure shows up on, and which specific faces. 

It would also be helpful to do exact same tests in same conditions using only production FW just to make sure this is related to new firmware.

Thanks!

bigjosh, 2021-07-15:
I compared the production version of the BIOS with the new 382 version and there do not seem to be any relevant changes around the IR communications stuff, so we are probably looking at something subtle and/or wierd! LMK if any things I can try on my side that seem likely to make this show up. Thanks!