bigjosh/Move38-Arduino-Platform #110: Issue installing Blinks Library via Arduino IDE

jbobrow, 2023-02-22:
Witnessed this issue in a couple of different ways. I believe this is the same error I witnessed while assisting people at the latest game jam. Using Arduino Legacy would allow folks to install as documented, but the Arduino 2.0 would show this error. I believe I witnessed this on mac computers and the following is the error code on a Linux machine. 

`Error during install: Cannot install platform: installing platform move38:avr@1.3.0: extracting archive: Create file /home/bga/.arduino15/tmp/package-647673701/Move38-Arduino-Platform/libraries/Examples05/BrokenRainbow/Broken Rainbow Map.png: open /home/bga/.arduino15/tmp/package-647673701/Move38-Arduino-Platform/libraries/Examples05/BrokenRainbow/Broken Rainbow Map.png: permission denied`

Source [here](https://forum.move38.com/t/problems-installing-blinks-sdk-with-arduino-cli/1276).

It appears that simply removing the image file could solve this issue. I think the diagram is great and belongs attached, so I am already hosting it [here](https://cdn.shopify.com/s/files/1/2127/8629/files/Broken_Rainbow_Map.png?v=1677034129), this way you can simply include it in the comments of the code.

bigjosh, 2023-02-23:
Hmmm... Seems like this error is coming from this line in the new Arduino CLI...

https://github.com/arduino/arduino-cli/blob/7115365e15d5f10bc7dd6083378c6e4e60c00dfa/arduino/resources/install.go#L66

...which is calling this line in an extractor library it uses...

https://github.com/codeclysm/extract/blob/40e27c6c8f3f10107c2d66cbcb8363fc938b8c17/extractor.go#L302

...which ends up failing here...

https://github.com/codeclysm/extract/blob/40e27c6c8f3f10107c2d66cbcb8363fc938b8c17/extractor.go#L331

bigjosh, 2023-02-23:
There do not seem to be any unexpected attribs on this file in the ZIP...

[image]

bigjosh, 2023-02-23:
Seems to install OK on Arduino 2..0.3 on Windows...

[image]

bigjosh, 2023-02-23:
Here is a new release to try to see if getting rid of the spaces in the file name changes anything...

https://github.com/bigjosh/Move38-Arduino-Platform/releases/tag/1.3.1

bigjosh, 2023-02-23:
Ah, I think this problem is that the directory is marked `RO` by Arduino and the extract function on Lunix applies this when the directory is created, so then the files in that directory can not be copied to, which is how the extract library does it. 

I removed the RO flag in the zip attached to release 1.3.1 which should fix the problem. Note that there is no commit for this release because git does not even track RO attribs!

closed 2023-02-23 by bigjosh

bigjosh, 2023-02-23:
I've updated the `boardsmanager.com`  [JSON file ](https://boardsmanager.com/package_move38.com-blinks_index.json)to point to the new version with this change.

To get the new Arduino to see the update, it look like you need to go into Preferences and delete the URL and then close the prefs window, open it again, and then add the URL back, and then close again.

bigjosh, 2023-02-23:
Yuck, looks like the new IDE is also incompatible with the (already hokey) libraries tree structure for examples. Fixed in commit 47fd3f1c7ca9e8c066bf5dc615d088ee9c341de7.