bigjosh/Move38-Arduino-Platform #112: please fix arduinoide board zip file (macos permission issue)

eddiegalvez, 2024-09-23:
c.f. https://forum.move38.com/t/problems-installing-blinks-sdk-with-arduino-cli/1276/10 the BrokenRainbow dir has a non user-read permission, preventing board install

bigjosh, 2024-09-25:
Is this related to https://github.com/bigjosh/Move38-Arduino-Platform/issues/110 ?

Which version of Arduino are you using?

eddiegalvez, 2024-09-25:
latest, aka 2.3.2. Near as I can tell you have not uploaded a fixed zip to the repo that carries the board, where the BrokenRainbow directory has no user-read permissions. My workaround was just to unzip, fix, re-zip it, host locally. I'll attach the corrected zip, but feel free to do the fix yourself rather than trust an uploaded zip file.
[Move38-Arduino-Platform-main.zip](https://github.com/user-attachments/files/17131736/Move38-Arduino-Platform-main.zip)

bigjosh, 2024-09-25:
Ok, I think I finally tracked down where that attrib is coming from and changed it in the file system, which should also hopefully change it in the zip file. 

Here is a new version of the zip file....
https://github.com/bigjosh/Move38-Arduino-Platform/releases/tag/1.3.3

Can you do me a favor and see if this one will install correctly on your platform?

Also, what platform are you on?

Thanks!

eddiegalvez, 2024-09-25:
macosx, 15.0. zip looks good!

bigjosh, 2024-09-26:
I have updated the board manager to point to this new zip file and made it be the current production release.

@eddiegalvez thank you for helping finally get this fixed!

closed 2024-09-26 by bigjosh